pathfinder-for-autonomous-navigation/psim #1: Develop of Robust UTC Model in PSim

kkrol27, 2019-09-05:
Currently, `psim` only supports a trivial model of time. At the start of every run, `truth.mission_time` defaults to zero and simply increments forward by `truth.dt` at every tilmestep. This simple model of time, however, is insufficient to support accurate conversions of vectors and coordinates from ECI to ECEF and vice versa.

To perform these conversions, we will need to leverage the MATLAB functions documented at the following link - I think the `ecef2lla`, `eci2lla`, `lla2ecef` and `lls2eci` functions may be the way to go: https://www.mathworks.com/help/aerotbx/axes-transformations.html?s_tid=CRUX_lftnav

We will also need the ECEF to ECI DCM to convert vectors between the coordinate systems. This should be possible given the `dcmeci2ecef` function at the following link:
https://www.mathworks.com/help/aerotbx/ug/dcmeci2ecef.html

The last thing we need to keep in mind is to ensure we have a way from converting from our UTC model to the proper GPS time that the satellite would see. This will be necessary both for verifying flight code in `sim` and for SHITL testing.

What "time" exactly is passed to the flight computer from Piksi, how are we representing it on the flight computer (we'll need this for the sim later), and do we have a proposed conversion from UTC to GPS time?

@stewartaslan @nhz2 @tanishqaggarwal

nhz2, 2019-09-05:
The Piksi passes GPS time to the flight computer. GPS time, unlike UTC doesn't have random leap seconds which is nice. For all of the models, time is a double or float in seconds since the start of some GPS week( currently set as week 2045 around march 17th 2019). UTC isn't very useful for precise physical modeling because of the leap seconds.

Technically to get the ECEF to ECI DCM, UT1, which is a measure of how much the earth has rotated, should be used instead of GPS or UTC time, but that level of precision is hopefully not needed, especially if we are just using a J8 gravity model, where the earth is assumed to be rotationally symmetric anyways. 

With the J8 model, the only important effect of the earth's rotation is in converting from velocities in an inertial frame to velocities in ECEF(which is what Piksi returns) because the of the  ω×r part of the velocity. For this ω is usually assumed to be constant.

tanishqaggarwal, 2019-09-05:
In Flight Software, the time is managed as follows:
- Stored in terms of milliseconds since OS bootup (this "system time" is provided to us by ChibiOS)
- The last GPS time received by Piksi is timestamped with system time, so that we can immediately get the current GPS time by just taking the last received GPS time and adding a delta-t.
- The absolute # of seconds since "the PAN Epoch" that Nathan mentioned is computed by subtracting the epoch from current GPS time.

I've already built C++ utilities for maintenance of these types of time on Flight Software, but it's not portable to desktop right now because I wrote it poorly. Rewriting it is a task on the docket.

I would recommend sending only GPS time to Flight Software from the simulation.

kkrol27, 2019-09-05:
@tanishqaggarwal Totally agree the best way forward is to only send GPS time to the FC from the sim - both simpler and is what happens on orbit.

@nhz2 What is the data type passed of the GPS time tho and how could we get that from UTC? I found this link (https://www.mathworks.com/matlabcentral/answers/446179-how-to-convert-utc-timestamp-to-unix-time-in-matlab-with-the-respect-of-leap-seconds) which accounts for leap seconds in UTC to POSIX time conversion (are GPS and POSIX time the same thing)?

As for time in the sim, I think we will want to make it as accurate as possible - ideally UTC to POSIX/GPS to ensure whatever model of time we place on the spacecraft is accurate enough as well. Could skipping leap seconds cause accuracy issues when converting magnetic field vectors from ECEF to ECI however?

I'm going to do a bit more reading tonight and try to structure the sim like this:
 * Provide a start time in UTC as `truth.start_time__utc = [year, month, day, hour, minute, second]`
 * Use the above to convert to POSIX/GPS time as `truth.start_time__gps`
 * Increment `truth.mission_time__ns` during the simulation by `truth.dt__ns` at each tilmestep
 * Use `truth.start_time__gps + truth.mission_time__ns * 1e-9`as current POSIX/GPS time at each timestep
 * Use current POSIX/GPS time to determine current UTC and give us all the conversions in the truth model that we need.

How does that sound? Current POSIX/GPS is what we would be sending to the satellite.

kkrol27, 2019-09-05:
@tanishqaggarwal: Not immediately, but when we start including C/C++ versions of GNC software we'll need to talk about where to put some of these time utility libraries and stuff. I don't think it's too important right now but just a heads up.

kkrol27, 2019-09-06:
@nhz2: Just to be clear, we're interested in this function call (`position = lla2eci(lla,utc)`) if our POSIX/GPS/Atomic to UTC conversion takes in to account leap seconds and then we'll forget about the UT1 issues?

tanishqaggarwal, 2019-09-06:
If you need GPS time for a GNC function, use a gps_time_t struct from here: https://github.com/pathfinder-for-autonomous-navigation/FlightSoftware/blob/tanishq/unit-tests/lib/utils/GPSTime.hpp

Otherwise, FSW can pass GNC a float time. FSW should take care of the conversions and GNC shouldn't have to worry

tanishqaggarwal, 2019-09-06:
The gps_time_t struct is currently unclean but I'll update you when it's a little better tested

kkrol27, 2019-09-06:
@tanishqaggarwal I agree and we will use a float for now. But really we would want to test that code in the sim without hardware potentially as well to verify the accuracy of all of our algorithms. I will take a look!

nhz2, 2019-09-06:
Double precision with units of seconds should be precise enough for time, and keeps everything in MKS.

kkrol27, 2019-10-08:
@nhz2 You think we can just call this closed due to #3?

closed 2019-10-09 by nhz2