pathfinder-for-autonomous-navigation/psim #9: Call C++ functions from matlab.

nhz2, 2019-10-07:
Learning how to call external c++ functions in Matlab. What kind of function wrappers are needed? Are namespaces supported?

kkrol27, 2019-10-07:
@nhz2 @stewartaslan I was planning on taking a look into this this week to start porting some things over. My starting point was going to be this link: https://www.mathworks.com/campaigns/offers/integrate-c-plus-plus-libraries-with-matlab.html?s_iid=fx_38964-example-matlab-class-wrapper-for-a-c-class_rcspot.

As I work through it I was going to make sure to keep in mind we'll want to easily be able to pass vectors, matrices, scalars, etc between the two programs. Hopefully we can develop a simple framework for it pretty quickly.

Any suggestions on an algorithm in PSim to start with?

stewartaslan, 2019-10-07:
Attitude controllers would be good, although you will have to dig into some quaternion stuff as well so that alone could be a process (such as defining quaternion operations and stuff)

nhz2, 2019-10-07:
There is also the sun vector, magnetic field, earth rotation, and J8 gravity models in [https://github.com/pathfinder-for-autonomous-navigation/Models/tree/master/include](url)

Right now the magnetic field model is very slow in Matlab.

kkrol27, 2019-10-07:
Sounds good - thanks. One thing we do need to start thinking about is how accurate we want our truth model to be and how accurate our models on the satellite would be. In an ideal world, we'd want the simulation to be able to verify approximation we make flight computer side are valid.

nhz2, 2019-10-19:
I tried using [https://www.mathworks.com/campaigns/offers/integrate-c-plus-plus-libraries-with-matlab.html?s_iid=fx_38964-example-matlab-class-wrapper-for-a-c-class_rcspot](https://www.mathworks.com/campaigns/offers/integrate-c-plus-plus-libraries-with-matlab.html?s_iid=fx_38964-example-matlab-class-wrapper-for-a-c-class_rcspot) but it was really slow. I found that just writing a custom wrapper was way faster. I used [https://www.mathworks.com/help/matlab/matlab_external/standalone-example.html](https://www.mathworks.com/help/matlab/matlab_external/standalone-example.html) as an example.

kkrol27, 2019-10-27:
After speaking on the phone with @nhz2 and @tanishqaggarwal, it was determined there would be two main functions that would be ported to c++, live in the PSim repo, and be called by FSW. They are a state estimation function (to filter/determine attitude and filter/determine spacecraft position and velocity). I'll be mocking up a rough outline for v1.0 of the estimation function and set it up so it's called from MATLAB.

kkrol27, 2019-11-04:
I think we've discussed this thoroughly and have a good way to interface between MATLAB and CXX. Closing this issue. See #66 for future developments.

closed 2019-11-04 by kkrol27