pathfinder-for-autonomous-navigation/psim #13: Vector standardization

tanishqaggarwal, 2019-10-09:
Putting this in psim, but this issue is really repository-wide.

As we head into transferring controllers into C++, I want to bring up the fact that we have several competing standards for representing vectors. Here are our options thus far and who uses them:
- `float*` or `double*`
- `std::array<float>` or `std::array<double>`
- Kyle's Tensor library

We need to decide: where should we have these things, and where? Here are some existing constraints:
- Flight Software serializers are designed to interface with `std::array`
- ADCS Software runs with Kyle's tensors
  - This is a little bit of an issue for when we integrate the State System (which exists on Flight Software right now) for the purposes of enabling communication between the ADCS Teensy and the desktop. We'll have to have an easy way to convert values that the simulation computer cares about from the Kyle-tensor representation to `std::array`
- MATLAB autocoders are built for C, so they produce float*'s

I'm opening this ticket so that we can collect issues that crop up related to vector standardization.

kkrol27, 2019-10-09:
@tanishqaggarwal Is FSW running of `cxx11` or `cxx14`? I thought I saw 14 in the platformio.ini but may have misread.

nhz2, 2019-10-10:
@kkrol27 Do you know if matlab supports calling C++ functions with std::array, pointers, or your tensor library as inputs?

tanishqaggarwal, 2019-10-14:
@kkrol27 C++14.

I think it's fine for algorithms to internally use lin, but using lin objects for FlightSoftware would require rewriting Serializers for these objects. Best to just use lin on the psim/algorithms side, or in existing places with ADCSSoftware.

Are there utilities for converting lin objects to std::array? This will be necessary when the state system is integrated into ADCSSoftware.

kkrol27, 2019-10-14:
They are all backed by `std::array` actually so it'd just be a matter of adding an extra set of getters.

This may not be the place to talk about it, by state system you mean the debugging console? Currently nothing in the ADCS struct is a `std::array` or `lin::Vector` because of volatility. Every array in the state struct is just a raw pointer to a statically allocated volatile array.

tanishqaggarwal, 2019-10-14:
Yep I mean the debugging console.

kkrol27, 2019-10-15:
@nhz2 @stewartaslan @tanishqaggarwal

Alrighty, so if any of you are curious and want to check out what the newer `lin` library looks like, I've finally put it up on my GitHub (https://github.com/kkrol27/lin). Take a look when you have time and let me know what you think. I'd strongly suggest looking at the two programs in `examples/*`, the files in `include/algorithms/*`, the files in `include/algorithms/inl/*`, and the unit tests in `test/*` as that should give you a pretty overview of the capabilities of the library. `include/vector/vector.hpp` and `include/matrix/matrix.hpp` contain most of the type definition that will be instantiated outside the library.

You can run the tests and examples pretty easily (on Mac at least). The examples are built with `make examples` and then the binaries get dumped in `bin/examples/*`. The tests require GTest to be installed with macports in `/opt/local/*` and are built and run via `make && make test`.

I will also be working on putting together an overview README to document the general as the stuff in `include/core/*` is mostly undocumented and can get messy at times.

Let me know what you think. I was going to add it to the PSim repo as a submodule or something like that so every other PAN repo could have access to it if everyone is good with it.

tanishqaggarwal, 2019-10-16:
Thanks for this! Would it be better to include in CommonSoftware, instead?

kkrol27, 2019-10-16:
I was planning on putting it in here so that it's easier for people to make local bug fixes if needed. All of the heavy development using the library will be done in this repo and FSW and ADCS Software will need to import algorithms from PSim already anyway.

kkrol27, 2019-10-16:
If @nhz2 and @saa243 think it looks good, I'll go ahead and add it to PSim.

nhz2, 2019-10-19:
For quaternion support, the Vector4f and Vector4d types are great. The additional quaternion functions we need are:

`utl_triad(const Vector3f& N_sun, const Vector3f& N_mag, const Vector3f&  B_sun, const Vector3f& B_mag, Vector4f& result)` 

`utl_quat_cross_mult(const Vector4f& q1, const Vector4f& q2, Vector4f& result)` 

`utl_quat_cross_mult(const Vector4d& q1, const Vector4d& q2, Vector4d& result)` 

`utl_dcm2quat(const Matrix3x3f, Vector4f& result)` 

`utl_quat_conj(const Vector4f& q, Vector4f& result)`

`utl_quat_conj(const Vector4d& q, Vector4d& result)`

`utl_rotateframe(const Vector4f& q, const Vector3f& v, Vector3f& result)`

`utl_rotateframe(const Vector4d& q, const Vector3d& v, Vector3d& result)`

These functions are currently in Matlab in the `MATLAB/utl` folder.

I'm not sure if you have this already, but having a way to cast a Vector4 to a Vector3, where the last element is discarded would be nice. 

For Vector3 math, we need cross product and dot product.

kkrol27, 2019-10-20:
I will open up another issue to handle quaternion support and move stuff over there. The utility functions can definitely start moving over to c++.

To answer your other questions:
 1. Current you could 'reference' (read only or read-write) the first three elements of a vector by writing something like this:

        lin::Vector4d quat = {0.0, 0.0, 0.0, 1.0};
        std::cout << lin::ref<3, 1>(quad, 0, 0) << std::endl;

    which would print out:

        0.0
        0.0
        0.0

    You can also take the reference and perform any other kind of operation, assignment, dot product, cross product (when that is supported) that you can do with a `lin::Vector3d`. To extract the first three elements as there own vector you can do something like:

        lin::ref<3, 1>(quad, 0, 0).eval(); // Produces a lin::Vector3d temporary

 2. Dot products are supported for all streams that are vectors - i.e. known to have exactly one column or row at compile time. It's just the dot function called as `lin::dot(a, b)` where `a` and `b` are vectors/row vectors in any combination - just need to have the same length. You could also, if you really wanted to calculate a dot product with the matrix multiplication operator:

        lin::Vector4d a, b;
        auto const x = (lin::transpose(a) * b)(0, 0);
        // ^^ Actually returns a 1 by 1 matrix so access the only element

kkrol27, 2019-10-20:
I go ahead and close this for now because I think generally things are figured out. I will also be getting together a quick start sorta overview for the library.

closed 2019-10-20 by kkrol27