pathfinder-for-autonomous-navigation/psim #19: Discretizing main.m

tanishqaggarwal, 2019-10-11:
Discretize `main.m` into three functions: `main_setup`, `main_loop`, and `main_teardown` so that the interface between the MATLAB simulation and the Python device interface is standardized.

tanishqaggarwal, 2019-10-11:
We'll actually need to split `main_loop` into probably two functions, `main_loop_pre_device_ix` and `main_loop_post_device_ix` so that we can insert Teensy I/O in the middle. Ideally `main_loop_pre_device_ix` is written so that it can safely run simultaneously with the Teensy I/O. We should try to make the interface between Teensy and the simulation data be as easy-to-understand as possible--right now there are a lot of structs floating around.

nhz2, 2019-10-12:
Here's what I'm thinking for how python would interface with matlab. Basically every thing is in the dictionary main_state, which is constructed with initialize_main_state. Then in the loop, sensor reading dictionaries are constructed, send to the teensys, and main_state is updated in matlab. Finally the actuator commands from the teensy modifies the main state.

~~~ python

#matlab initialization
import matlab.engine
eng= matlab.engine.start_matlab()
eng.addpath('MATLAB', nargout=0)
eng.config(nargout=0)  # Initialize const
main_state= eng.initialize_main_state(seed,'not_detumbled')

then in the loop do
   #sense truth
   sensor_readings_follower = eng.sensor_reading(main_state['follower'],main_state['leader'])
   sensor_readings_leader = eng.sensor_reading(main_state['leader'],main_state['follower'])
   #update dynamics
   main_state = eng.main_state_update(main_state)
   send sensor readings to flight computers
   read actuator commands from flight computers
   #command actuators
   main_state['follower'] = eng.actuator_command(actuator_commands_follower,main_state['follower'])
   main_state['leader'] = eng.actuator_command(actuator_commands_leader,main_state['leader'])
~~~

What do you think?

tanishqaggarwal, 2019-10-12:
This looks great! I'll update the Python script with this once I'm able to integrate my pull request.

tanishqaggarwal, 2019-10-12:
One thing thought is that we need to be able to control loop timing, so we still need the mission_time and dt variables.

tanishqaggarwal, 2019-10-16:
Completed in #24 .