pathfinder-for-autonomous-navigation/psim #22: sensor_reading

nhz2, 2019-10-15:
Outputs the sensor readings of one satellite.

nhz2, 2019-10-16:
To test the control algorithms its important to see if they still work with sensor noise and bias. For attitude control this is mainly gyro bias and noise and magnetometer bias and noise.

Here is some data recorded outside from the two magnetometers. The magnetometers will probably have additional noise and bias from the wheels, magnetorquers, and docking magnets.

The sensor noise models don't have to be super realistic, they just have to be worse than the real worst case noise/bias.

nhz2, 2019-10-16:
Also the sunsensor model and when the sunsensors can't see the sun.

kkrol27, 2019-10-16:
In general when there isn't sufficient lighting to detect the sun the ADCS will just return a flag that indicates the sun isn't in view. It only includes a photodiode in the sun vector calculation if the observed voltage is greater than 50% of the maximum expected intensity.

Would this sort of model work in PSim?

nhz2, 2019-10-16:
Yeah, just modify sensor_reading.m

Right now sun sensor calculations are done starting at line 30

I am putting the sun sensor normals in `my_satellite_state.sensors.sunsensor_real_normals` and `my_satellite_state.sensors.sunsensor_measured_normals`: These are size (3,20) matrixes initialized in `initialize_main_state.m`. Should these should have units of volts instead of unitless? Right now they are just the ideal unit vectors. To get the true sun vector, you can call `get_truth('sat2sun body',my_satellite_state.dynamics)`, and implement that functionality in `get_truth.m`, or for now, since `get_truth.m` doesn't exist yet, you can just use the quaternions and `env_sun_vector.m`.

kkrol27, 2019-11-24:
@nhz2 Do we want to keep this open or close it in favor of making multiple, more specific issues - like #79?

tanishqaggarwal, 2020-03-11:
Very old

closed 2020-03-11 by tanishqaggarwal