pathfinder-for-autonomous-navigation/psim #47: Code J4/J8 Orbit Propagator in C++

kkrol27, 2019-10-21:
We need to code a J4/J8 orbit propagator in C++ to fly on the spacecraft. This will required an integrator implementation like `ode45` to be included in PSim as well.

tanishqaggarwal, 2019-10-22:
Along with this issue (since it directly relates to FSW): we need to figure out a painless way to share code between psim and FlightSoftware.

I think importing psim as a library into FlightSoftware would be ideal, but this requires being cognizant about the fact that platformio only finds library files if they're in the root-level of src/ and include/. This might be tricky (or not) given that `lin` has already been placed in a place that PlatformIO might not be able to find it--have you been able to test this or know about the inclusion structure?

The other option would be to add psim as a submodule to FlightSoftware. I'm not a fan of this because it makes the directory structure even more convoluted and difficult to import into PlatformIO. Unless I'm imagining a difficulty where it doesn't actually exist.

I think a first step would be to create stub files for things we anticipate absolutely needing in FSW, and experimenting with FlightSoftware and platformio includes until everything flows. What do you think?

tanishqaggarwal, 2019-10-22:
Nevermind. I forgot that we're already tracking this integration on #41 .

tanishqaggarwal, 2020-03-11:
Duplicate of #78

closed 2020-03-11 by tanishqaggarwal