pathfinder-for-autonomous-navigation/psim #52: utl_vect_rot2quat C++ version and unit tests (including NaN test cases)

nhz2, 2019-10-27:
function qout = utl_vect_rot2quat(r1,r2)
%UTL_VECT_ROT2QUAT Returns a unit quaternion that will rotate r2 to r1 with minimal angle.
%   If r1 and r2 are very close to anti parallel, [0;0;1;0] is default.
%     Quaternions have the forth component as scalar.
%     r1 and r2 must unit vectors or have NaN or +- inf

kkrol27, 2019-10-27:
Actually have this done on a branch right now - however they don't check for NaN stuff. Do we really want the responsibility of checking whether or not something is NaN to a lower level utility function?

nhz2, 2019-10-27:
Yeah, low level utility functions should usually not need to check for NaN, propagation is automatic. Just make sure the code still runs if it has NaN input, and that the quaternion output becomes nonfinite if any input is nonfinite. I'll add that to the Matlab function documentation.

tanishqaggarwal, 2020-03-11:
@kkrol27: closing because the functions are already implemented, the NaN tests weren't done but we're using them at a FlightSoftware level anyway, so we have enough protection.

closed 2020-03-11 by tanishqaggarwal