pathfinder-for-autonomous-navigation/psim #65: Gravity Model Environment Function in CXX

kkrol27, 2019-11-04:
The last environment model that needs to be added in Earth's gravity model. We'll stub in a simple Newtonian gravity model initially to get state estimation up and running.

In the long term, we'll be coding a J4/J8 gravity model to use in an orbit propegator.

nhz2, 2019-11-17:
I found a C++ library for doing spherical harmonics. [https://geographiclib.sourceforge.io/](https://geographiclib.sourceforge.io/). This library uses full double precision. I'm going to try copy most of the code from that library into a simpler header only library that does an about 20x20 model, and see what calculations can be done in single precision.

nhz2, 2019-11-22:
I put all of the files in the `geograv` directory in `psim` temporarily to help debugging.

The main gravity function and the definition of the `Coeff` class is in `geograv.hpp`

`EGM96.hpp` has the coefficients for the 360x360 egm96 model.

`teensy_test` is an arduino sketch to run on the teensy.

`gravmodelgen.py` is a python script to convert text files with coefficients to the c++ header files with coefficients. 

`main.cpp` is a the test file for running on desktop.
## Installation

First install [geographiclib](https://geographiclib.sourceforge.io/html/index.html):
On mac you can just `brew install geographiclib`
Then run `/usr/local/sbin/geographiclib-get-gravity all` to install all of the gravity models.

 To run the tests,
First open terminal in the `geograv` directory, then run the following commands:
`mkdir build`
`cd build`
`cmake ..`
`make`
`./TestGeoGrav`

To generate new coefficient header files, use `gravmodelgen.py`, for example:
`python gravmodelgen.py -f egm96 -n 360 -o EGM96.hpp -r 6378137.0 -u 0.3986004418e15`
and
`python gravmodelgen.py -f egm96 -n 20 -o EGM96_20.hpp -r 6378137.0 -u 0.3986004418e15`

closed 2020-02-06 by nhz2