pathfinder-for-autonomous-navigation/psim #78: Add GPS State Estimation Function (1.0)

kkrol27, 2019-11-08:
Add a CXX GPS state estimation function for GNC version 1.0.

In the far field, this function will provide position and velocity measurements for the leader and follower spacecrafts in ECI. We will use a fixed step size integrator to propagate both orbits during each cycle of the flight computer.

If a timestamped position, velocity of the other satellite is received, it will be propagated forward to the present time with a variable step size integrator and replace the old position, velocity measurement of the old spacecraft.

In the near field, we will also need to provide a relative position, velocity in the LVLH/Hill frame. Here we will add a simple state filter based on (most likely) Clohessy-Wiltshire dynamics to get a better relative velocity measurement. (@stewartaslan will open a separate issue for this filter - <link here>).

nhz2, 2019-12-08:
TODO, figure out what to do with float vs fixed RTK data

nhz2, 2020-02-13:
The estimator should also estimate the energy of both orbits.

nhz2, 2020-02-26:
The estimator should respond to / predict maneuvers.

nhz2, 2020-03-29:
Some issues with the current MATLAB orbit estimator implementation.
1. The orbit estimator can call the gravity function an unlimited number of times, this isn't possible on a teensy.
2. The estimator is numerically unstable.

To solve these I'm working on a redesigned orbit estimator, where the number of calls to the gravity function is controlled. Also instead of a giant EKF the new filter will be multiple filters running in parallel, each taking in different combinations of sensor readings/ ground data. That way if one EKF fails because of a bad sensor, the other filters can be used.

PR #297 (2021-02-09): Orbit Library Orbit Estimator Implementation
Closes #78.

### Summary of changes

- Small tweaks to the `//:gnc` target to avoid loosing headers in nested directories within `include/orb/`.
- Square root extended Kalman filter orbital state estimator implementation. GPS position and velocity readouts are taken as sensor measurements in ECEF.
- Flight computer orbit estimator model for `psim` with plots config included.
- Simple orbit estimator simulation for testing in `psim` standalone.

### Testing

See the following plots generate below showing valid estimates of position and velocity over  [... cut]
closed 2021-02-09 by kylekrol