pathfinder-for-autonomous-navigation/psim #229: Adjust Attitude Estimator Sun Vector Error Representation

kkrol27, 2020-05-15:
Currently, we define our expected sun vector distribution as a spherical coordinate representation plus or minus a predetermined uncertainty in `phi` and `theta`. While the sun vector is around the equator this is alright as a given step in `phi` or `theta` traverse a similar distance on the unit sphere. If the expected sun vector, however, is near a pole (plus or minus z), this representation is no longer consistent.

We need to define a local error distribution around the expected sun vector for this to work more consistently.

shihaocao, 2020-12-06:
[image]

shihaocao, 2020-12-06:
[image]

shihaocao, 2020-12-06:
First time scale was on the order of day : hours : mins

PR #289 (2021-01-15): Fix Attitude Estimator Sun Vector Representation

# Fix Attitude Estimator Sun Vector Representation

Fixes #229.

### Summary of changes

The attitude estimator was having difficulty in processing sun vectors when the expected measurement was nearly aligned with the body z axis. Given the filter used a spherical angle sun vector representation, some of the generated sigma points would have `phi` values nearly 180 degrees apart from one another (if a sigma point rotation tipped the expected sun vector measurement over top of the z axis). This caused the filter to have severe stability issues.
 [... cut]
closed 2021-01-15 by kylekrol