pathfinder-for-autonomous-navigation/psim #257: Failing builds on MacOS

kkrol27, 2020-10-20:
@tanishqaggarwal reported this build issue on MacOS:
```
(venv) ➜  psim git:(master) bazel test //test/psim:all
INFO: Analyzed 2 targets (14 packages loaded, 12607 targets configured).
INFO: Found 2 test targets...
ERROR: /Users/tanishqaggarwal/Documents/pan/repositories/psim/BUILD.bazel:55:26: C++ compilation of rule '//:psim_truth' failed (Exit 1) wrapped_clang failed: error executing command external/local_config_cc/wrapped_clang '-D_FORTIFY_SOURCE=1' -fstack-protector -fcolor-diagnostics -Wall -Wthread-safety -Wself-assign -fno-omit-frame-pointer -O0 -DDEBUG '-std=c++11' -iquote . -iquote ... (remaining 38 argument(s) skipped)
Use --sandbox_debug to see verbose messages from the sandbox
In file included from src/psim/truth/time.cpp:25:
In file included from include/psim/truth/time.hpp:32:
bazel-out/darwin-fastbuild/bin/include/psim/truth/time.yml.hpp:45:58: error: use 'template' keyword to treat 'get' as a dependent template name
    prefix_t_ns(_prefix+".t.ns", config[_prefix+".t.ns"].get<Integer>()),
                                                         ^
bazel-out/darwin-fastbuild/bin/include/psim/truth/time.yml.hpp:47:61: error: use 'template' keyword to treat 'get' as a dependent template name
    prefix_dt_ns(_prefix+".dt.ns", config[_prefix+".dt.ns"].get<Integer>()),
                                                            ^
2 errors generated.
INFO: Elapsed time: 41.799s, Critical Path: 17.27s
INFO: 46 processes: 46 darwin-sandbox.
FAILED: Build did NOT complete successfully
//test/psim:core_test                                                 NO STATUS
//test/psim:truth_test                                          FAILED TO BUILD
FAILED: Build did NOT complete successfully
```
Going to update the calls to `get` to keep apple clang happy for the time being.

tanishqaggarwal, 2020-10-20:
Also seeing the following issue when building GNC code with `pio test -e native`

```
(venv) ➜  psim git:(master) pio test -e native -v
Collected 6 items

Processing attitude_estimator in native environment
------------------------------------------------------------------------------------------------------------------
Building...
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain+, Compatibility ~ soft
Found 1 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <lin> 0.0.1 (/Users/tanishqaggarwal/Documents/pan/repositories/psim/lib/lin)
Building in release mode
ar rc .pio/build/native/libda8/liblin.a
ar: no archive members specified
usage:  ar -d [-TLsv] archive file ...
	ar -m [-TLsv] archive file ...
	ar -m [-abiTLsv] position archive file ...
	ar -p [-TLsv] archive [file ...]
	ar -q [-cTLsv] archive file ...
	ar -r [-cuTLsv] archive file ...
	ar -r [-abciuTLsv] position archive file ...
	ar -t [-TLsv] archive [file ...]
	ar -x [-ouTLsv] archive [file ...]
*** [.pio/build/native/libda8/liblin.a] Error 1
=========================================== [FAILED] Took 1.06 seconds ===========================================

Processing groundpropagator in native environment
------------------------------------------------------------------------------------------------------------------
Building...
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain+, Compatibility ~ soft
Found 1 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <lin> 0.0.1 (/Users/tanishqaggarwal/Documents/pan/repositories/psim/lib/lin)
Building in release mode
ar rc .pio/build/native/libda8/liblin.a
ar: no archive members specified
usage:  ar -d [-TLsv] archive file ...
	ar -m [-TLsv] archive file ...
	ar -m [-abiTLsv] position archive file ...
	ar -p [-TLsv] archive [file ...]
	ar -q [-cTLsv] archive file ...
	ar -r [-cuTLsv] archive file ...
	ar -r [-abciuTLsv] position archive file ...
	ar -t [-TLsv] archive [file ...]
	ar -x [-ouTLsv] archive [file ...]
*** [.pio/build/native/libda8/liblin.a] Error 1
=========================================== [FAILED] Took 1.18 seconds ===========================================

Processing kalman_utl in native environment
------------------------------------------------------------------------------------------------------------------
Building...
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain+, Compatibility ~ soft
Found 1 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <lin> 0.0.1 (/Users/tanishqaggarwal/Documents/pan/repositories/psim/lib/lin)
Building in release mode
ar rc .pio/build/native/libda8/liblin.a
ar: no archive members specified
usage:  ar -d [-TLsv] archive file ...
	ar -m [-TLsv] archive file ...
	ar -m [-abiTLsv] position archive file ...
	ar -p [-TLsv] archive [file ...]
[... 95 lines cut ...]
kalman_utl          github         IGNORED
kalman_utl          teensy35       IGNORED
kalman_utl          teensy35_ci    IGNORED
kalman_utl          teensy36       IGNORED
kalman_utl          teensy36_ci    IGNORED
misc                native         FAILED    00:00:01.140
misc                github         IGNORED
misc                teensy35       IGNORED
misc                teensy35_ci    IGNORED
misc                teensy36       IGNORED
misc                teensy36_ci    IGNORED
orbit               native         FAILED    00:00:01.119
orbit               github         IGNORED
orbit               teensy35       IGNORED
orbit               teensy35_ci    IGNORED
orbit               teensy36       IGNORED
orbit               teensy36_ci    IGNORED
orbit_controller    native         FAILED    00:00:01.086
orbit_controller    github         IGNORED
orbit_controller    teensy35       IGNORED
orbit_controller    teensy35_ci    IGNORED
orbit_controller    teensy36       IGNORED
orbit_controller    teensy36_ci    IGNORED
===================================== 6 failed, 0 succeeded in 00:00:06.720 =====================================
```

kkrol27, 2020-10-20:
@tanishqaggarwal I think the above may actually require updates to the library.json file in `lin`. @shihaocao I think fixed the psim one recently.

PR #259 (2020-10-22): Fix Builds on MacOS
Fixes #257.

## Summary of Changes

* Update lin pointer to include compatibility fixes.
* Include the `template` keyword on calls to `get<...>`, `cast<...>`, et cetera.
closed 2020-10-22 by kkrol27