pathfinder-for-autonomous-navigation/psim #281: Verify and Update GNC Documentation with Regard to Controller Torque Sign

kkrol27, 2020-12-06:
@shihaocao and I spent a long time chasing down a potential sign error across flight software, psim, gnc, and adcs code. We shoudl audit the controllers implemented in gnc and verify all signs and comments are correct.

shihaocao, 2020-12-06:
This was discovered because Flight Software required a -1*<the output of GNC controllers> in order to show that the space craft could detumble with working with C++ psim

shihaocao, 2020-12-06:
https://github.com/pathfinder-for-autonomous-navigation/psim/blob/3f837a3f6e8d61a5fc9b1d6bce612ee312bfaaed/src/gnc/attitude_controller.cpp#L157

shihaocao, 2020-12-06:
The above line of code is correct assuming the following:

- Error is defined as a positive number if our current value is mathematically greater than our target value
- Kp and Kd are positive values (they are)
- And lastly, GNC is outputing a torque that should be delivered to the satellite body itself in order to hit a target objective, and 0 angular rate). This is the desired scheme per @kkrol27 

If the line of code is correct that also implies the following:
- Comments in GNC need to be updated to properly reflect that it is outputing a desired torque to the spacecraft frame in the body frame.
- FSW needs to explicitly negate the output of GNC controllers in order to forward a desired torque delivered to the wheels, in order to cause the negative of the negated command to be delivered to the spacecraft frame.