pathfinder-for-autonomous-navigation/psim #351: NASA Monte Carlo

govindchari, 2022-01-05:
We need to report coordinates in J2000 MEME Format

PSim propagates in ECEF without precession terms, so it is slightly inaccurate, but is fine to propagate with for a week

However, we need to reinitialize a few parameters (quat_ecef0_eci and precession rate) with the values from the current day

There is a matlab script in the MATLAB/ folder titled config.m that does this. Just put in the date on line 35 and the parameters are computed in lines 62 and 64.

There is also a PR open that will allow these inputs to be put into the config file of psim

When running the Monte Carlo, we should also reseed the config file

nhz2, 2022-01-06:
Take a look at https://github.com/pathfinder-for-autonomous-navigation/psim/blob/13824dfb94d07a672ce1e97e89e205b815bf1584/MATLAB/environmental_models/helper_functions/eci_ecef_conversions.py if you want an example of converting between ECEF and ECI with astropy and GPS time. The eci ecef stuff in `MATLAB/config.m` is just a rough conversion that is good enough for attitude estimation and control, but has some issues.

Also, the date on line 35 of `MATLAB/config.m` is the date when the earth was at perihelion in 2019, its used for the simplified sun vector model. http://www.astropixels.com/ephemeris/perap2001.html has some other perihelion dates.

Change the initial GPS week number:
https://github.com/pathfinder-for-autonomous-navigation/psim/blob/13824dfb94d07a672ce1e97e89e205b815bf1584/MATLAB/config.m#L14
To update `const.INIT_DYEAR`which is used in the magnetic field model:
https://github.com/pathfinder-for-autonomous-navigation/psim/blob/13824dfb94d07a672ce1e97e89e205b815bf1584/MATLAB/config.m#L16

`const.tp_earth` which is used in the sun vector model:
https://github.com/pathfinder-for-autonomous-navigation/psim/blob/13824dfb94d07a672ce1e97e89e205b815bf1584/MATLAB/config.m#L36

And the other parameters used in the rough earth rotation model.
https://github.com/pathfinder-for-autonomous-navigation/psim/blob/13824dfb94d07a672ce1e97e89e205b815bf1584/MATLAB/config.m#L57-L65

shihaocao, 2022-01-07:
@nhz2 @govindchari Is it possible to just run the python script that Nathan linked instead of the Matlab script? The ground server will not be able to easily run MatLab scripts.

shihaocao, 2022-01-08:
https://docs.astropy.org/en/stable/time/index.html#convert-time-scale
Use this for time dump into utc

shihaocao, 2022-01-08:
TODO folder login dump